diepala/cantor #2: QMK firmware throws error on build

Ansimorph, 2022-03-06:
When using your fork of qmk-firmware and the qmk cli on OSX (12.2.1) I get the following error when compiling the included default keymap:

```
$ bg:~ $ qmk compile -kb cantor -km default
Ψ Compiling keymap with gmake --jobs=1 cantor:default

QMK Firmware 0.16.0
Making cantor with keymap default

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

☒ Matrix data out of bounds for layout LAYOUT_split_3x6_3 at index 6 (k4A): 4, 0
gmake[1]: *** [builddefs/build_keyboard.mk:335: .build/obj_cantor/src/layouts.h] Error 1
gmake: *** [Makefile:413: cantor:default] Error 1
Make finished with errors
```

diepala, 2022-03-06:
You are right, there was an error with the QMK firmware in the PR. It should be now solved.

The firmware has now been moved to the cantor branch of my forked qmk repository: https://github.com/diepala/qmk_firmware/tree/cantor

Can you check if it works now?

Ansimorph, 2022-03-07:
It compiles fine. Thanks so much for the quick fix!

closed 2022-03-07 by Ansimorph