diepala/cantor #3: Add Cantor to splitKbCompare

azzamsa, 2022-03-21:
Hi.

Would you like to  add this to https://github.com/jhelvy/splitKbCompare :)

Would like to compare to other keyboard.

diepala, 2022-03-21:
I will take a look on how to do it when I have time. Meanwhile, you can see how the switches are layed out with kiCad.

azzamsa, 2022-05-23:
Related: https://github.com/jhelvy/splitKbCompare/issues/84

azzamsa, 2022-05-26:
In case anyone want to print the layout (without installing the KiCad):

[image]

[cantor.pdf](https://github.com/diepala/cantor/files/8779709/cantor.pdf)

l4u, 2022-06-27:
It's available at splitKbCompare now.

azzamsa, 2022-10-24:
> It's available at splitKbCompare now.

Thank you!

closed 2022-10-24 by azzamsa