diepala/cantor #6: Create a MX compatible version

twalker, 2022-04-04:
It'd be fantastic to have a version of Cantor that was compatible with MX switches. 

MX compatible hot swap sockets would be ideal IMHO, they:
* Are easier to solder than switches themselves
* Allows exploring different switches

twalker, 2022-04-07:
> Have you tested or seen other plateless keyboards with the configuration you propose?

No, I have not. All the MX compatible keyboards I've built (Corne, Kyria, Sofle, etc) used a base and switch plate. They're quite stable with a case, but I imagine they'd be less stable with one. I'm naively imagining a different version of the PCB with the same layout, with non-choc key spacing, along with a switch and base plate designs. If that conflicts with the goals (low cost) of the Cantor, or is impractical, then I totally understand.

diepala, 2022-04-09:
I don't want a switch plate to be mandatory for the build. The MX keyboard won't be a priority for now, but in the future we could make some tests for MX switches.

twalker, 2022-04-09:
@diepala , I look forward to any future possibile explorations of an MX version. As a software engineer, I'm not very skilled with electronics or PCB design, but would happy to be a tester if/when one is needed. Thank you.

ratsclub, 2022-06-24:
Hmm, I just stumbled upon this issue. I can't really build this keyboard for an affordable price because of its switch... There are some MX compatible switches from Red Dragon that are **really** affordable tho. So I'd be happy to try the MX model if it happens to come out!

taurhine, 2022-08-17:
This is a great improvement over corne design. It would be awesome to have it with MX switches!

artsbentley, 2022-09-14:
Yes would love an MX design!

diepala, 2022-09-16:
Seeing that many people would like this version, I will probably make an MX version.

fabikroeger, 2023-02-14:
An MX design would be awesome :+1:

taurhine, 2023-02-18:
or MX Low Profile :heart_eyes: https://www.cherrymx.de/cherry-mx/mx-low-profile/mx-low-profile-red.html

mejobloggs, 2023-02-27:
Redragon AliExpress has two MX compatible low profile switches... Could be a potentially good option!

https://www.aliexpress.com/item/1005004299855522.html

https://www.aliexpress.com/item/1005003606992057.html

However, I have not used either, so I can't verify if they genuinely are MX compatible

albertothedev, 2023-04-04:
Any progress on this? I really like the improvements made over CRKBD but the use of Choc keys is a dealbreaker for me, like I'm sure it is for many others.

JayceFayne, 2023-12-28:
This issue was fixed by #25

closed 2024-01-05 by diepala