diepala/cantor #7: Vial Support

DrewTChrist, 2022-04-06:
I'd like to add Vial support for this keyboard. I've already got it working from my local machine and my Cantors. This would require a fork of vial-qmk and merging it back into their main. I could handle that, but since you designed the keyboard I thought it would be more appropriate if you forked vial-qmk and then I could make a PR to your fork and then it could be merged back into vial-qmk main.

What are your thoughts?

wvffle, 2022-04-08:
The beauty of open source software is that anyone can contribute. You should do it yourself in my opinion ;)

DrewTChrist, 2022-04-09:
I went ahead and made the PR like @wvffle suggested. 

That can be tracked [here](https://github.com/vial-kb/vial-qmk/pull/141). Also my fork is [here](https://github.com/DrewTChrist/vial-qmk) if anyone would like to try it. 

It builds just the same as qmk, but when you run Vial it should be discoverable and the keymap should be editable from the gui. The two outer top corner keys should "unlock" it.

diepala, 2022-04-09:
This is great! I haven't used vial before. Thanks for making the PR to add the vial support.

DrewTChrist, 2022-04-09:
Happy to help! It looks like my branch was merged into their main branch. Cheers

closed 2022-04-09 by DrewTChrist