diepala/cantor #10: Wireless technically possible?

kekscode, 2022-04-24:
Hello 👋
Long time corne low profile user here. I recently built a corne with ZMK firmware and nice!nano v2 MCU and don’t want to go back having canbles.

I love your pcb design (especially the stagger) and the diodeless approach. Do you think it is possible build a wireless variant of the cantor?

diepala, 2022-04-28:
You would need a blackpill with bluetooth support, but I don't know if that exists.

closed 2022-04-28 by diepala

kekscode, 2022-04-28:
Maybe not (yet?). I‘ll build a cantor this weekend and if it suits me i will do some research. Kudos for the simple design. I am looking at the pcbs right now and i like the KISS approach a lot 👍

azzamsa, 2022-05-16:
@kekscode any update? I love to have the wireless version too.

kekscode, 2022-05-17:
Not really feasible at the moment.

funkycode, 2023-03-07:
I would assume it is with https://mechwild.com/product/pillbug/
Blackpill compatible with BLE and ZMK support