diepala/cantor #15: Doesn't work after booting into the OS

vanarok, 2022-08-10:
Cantor does not work if connect while booting into the OS. I need to reconnect it to make it work. I have another custom keyboard - corne and it does not have this problem. Any ideas?

bennypowers, 2024-01-23:
I have this same issue with the qmk firmware. Would appreciate some guidance in debugging

j-hap, 2024-08-04:
adding `#define SPLIT_WATCHDOG_ENABLE` in the `config.h` fixed the cold boot issue for me. Credit to Dasky on the QMK Discord.

closed 2024-08-04 by vanarok

bennypowers, 2024-08-04:
for those following along from home, when I added that line to the config.h in the cantor dir, it broke my left half. I fixed it by pluggin in a spare keyboard and moving that line to the config.h in the keymap directory, then reflashing both sides with boot0-nrst acc'd to readme.