diepala/cantor #16: Only USB-C ports

Lumi317, 2022-10-07:
Would it be possible to create a version of the Cantor whose TRRS ports are replaced by USB-C ports?

diepala, 2022-10-16:
Using USB-C ports was an option I thought about when designing the Cantor, but I discarded it because soldering would be more difficult to do by hand. So it is not going to be implemented.

closed 2022-10-16 by diepala

Lumi317, 2022-10-16:
Thank you for taking the time to answer me and thank you very much for creating and sharing the Cantor. <3

Lumi317, 2022-10-17:
I may be wrong, but I seem to have read that some complicated soldering can be done by machines in factories producing PCBs if they are provided with a PCBA file.
Wouldn't this be a way to take advantage of the solidity and universality of USB-C on the Cantor while avoiding the difficulties of manual soldering?