diepala/cantor #17: Setting master side

arielarmijo, 2022-10-22:
I want to define the master side defining the SPLIT_HAND_PIN [as the documentation says](https://docs.qmk.fm/#/config_options?id=defines-for-handedness).

Is this possible? and which pin of the microcontroller can be used to determine the master side?

best regards!

diepala, 2022-11-19:
WHen designing the keyboard, I did not consider any pin to define the split hand. You can change the master side by firmware, including or removing the `#define MASTER_RIGHT`, or you might use one of the blackpill free pins, depending on what you want to achieve.

222Phoenix, 2025-08-17:
A few years late on this, but if you were looking for a way to plug in either side and have that be the master, then I got it to work with EE_HANDS (https://docs.qmk.fm/features/split_keyboard#handedness-by-eeprom). 

qmk flash -kb cantor -km <your keymap folder> -bl dfu-util-split-right
qmk flash -kb cantor -km <your keymap folder> -bl dfu-util-split-left

Alternatively you can set up swap hands and then trigger it with a macro or combo. I was doing that for a long time but now that EE_HANDS works I don't need it.