diepala/cantor #19: 4 keys not working

usizu, 2022-11-01:
Just built this great keyboard, have an issue with 4 keys not working on the right-side though.

They correspond on the default mapping to the `[`, `n`, `m` and `,` keys. So the top-right key, and the bottom left-three (within the main block).

QMK key tester:
[image]

I suspect the issue is with the controller, not the switches. If I swap the controller with the one on the other side, then the same mirror-positions stop working on that side.

I'm new to keyboards and hope to get some help in troubleshooting this.

1) Could it be something simple like one or more improperly soldered pins on the controller?
2) Is it easy to determine which pins govern these keys, so I can adjust the soldering on those pins?
3) What would be sufficient criteria to determine if I should just buy a new blackpill?

Thanks for any help.

diepala, 2022-11-19:
Hi, probably you will have to replace blackpill, but you can check if the balckpill is working by short-circuiting to ground, with a wire, the pins corresponding to the non-working keys. Connecting a pin to ground is equivalent to pressing the associated key, so you should see as if that key is pressed in the computer.

Your keys correspond to pins B1, A6, A7 and B0.

usizu, 2022-12-10:
Hi, thanks for this message.
I finally got around to trying this, and indeed the pins from A6 up to B1 don't register a keypress whereas the others do, so this seems to affirm that its this blackpill unit that is faulty, thanks for the help

closed 2023-02-23 by diepala