diepala/cantor #28: MCU Orientation

fla-t, 2023-11-05:
I have ordered a cantor PCB, and just waiting for the switches to be built. I noticed that you have heavily said that the controller must be face up. 

But, in the PCB I noticed that there are 2 Pin arrays, It seems like they are mirror opposite of each other. Meaning I don't need to face down the PCB if I choose the one that has a square white mask around it.

My question is am I right? 

Sorry if it doesn't feel like an issue, because I have no other way to ask this.

diepala, 2023-11-05:
As it was originally designed, the microcontroller should be soldered on the white mask face down as shown:

[image]

But you are right in that it can be soldered face up on the other row of holes.

closed 2023-11-05 by diepala

fla-t, 2023-11-06:
Thank you so much, I think I should update the docs a little bit.