diepala/cantor #29: The MX release contains two gerber files.

The-Abe, 2024-03-07:
The zip release for the MX version contains the gerber files and another zip containing gerber files.

This causes jlcpcb to reject the board during review.

Blubberland, 2024-04-08:
Cannot confirm.
The second layout was automatically stripped away when I added it to the cart. (I was wasn't sure how it would be handled during the order process, so I just waited until delivery).
Package was delivered today and contains 5 pieces (not 10).
Would have been nice to know beforehand. I would have ordered an even number. Could be added to the [ordering guide](doc/pcb_ordering_guide.md).

simonvanbernem, 2024-04-26:
I encountered the same thing.

jlcpcb emailed me, asking what to do about the internal zip file, and I instructed them to ignore it (which I hope was the right thing, as it claims to be v1.61 dated 10-24-2023, while the release zip is v1.6 from 11-05-2023 (later date, but earlier version??).

It would be great if the internal zip file could be deleted, so that newbies like me don't get surprised by this.

yantongl, 2024-05-16:
I ran into the same problem.

If you unzip the `gerber-cantor-mx-rev1.6.zip` file, you will see it contains another zip file `cantor_mx_v1.61_10-24-2023.zip`. I am going to tell them to ignore the internal zip file as @simonvanbernem said above.

[image]

simonvanbernem, 2025-04-24:
> If you unzip the `gerber-cantor-mx-rev1.6.zip` file, you will see it contains another zip file `cantor_mx_v1.61_10-24-2023.zip`. I am going to tell Jlcpcb to ignore the internal zip file as [@simonvanbernem](https://github.com/simonvanbernem) said above.

For the record: This turned out to be the correct choice. Everything worked out in my case.