diepala/cantor #32: Keyboard doesn't work until I reconnect it

arielarmijo, 2024-09-05:
The keyboard works fine, I love it, but after I turn on the PC, the  keyboard won't work until I disconnect and re-connect it again.

I have build it with the blackpill [STM32F401CCU6 3.0](https://es.aliexpress.com/item/4001049840288.html). In my config.h file i have this code:
```
#pragma once

#define MASTER_LEFT
#define USB_WAIT_FOR_ENUMERATION
#define LAYER_STATE_8BIT
```

Is there any specific settings I should use to fix this issue?

Thanks for design this awsome keyboard. Here is my [build in reddit].(https://www.reddit.com/r/ErgoMechKeyboards/comments/yaotqf/my_cantor_build/).

dimipage, 2024-10-12:
Do you use a usb hub by any chance? I had the same issue and switching the hub or connecting directly to pc helped.

arielarmijo, 2024-11-11:
> Do you use a usb hub by any chance? I had the same issue and switching the hub or connecting directly to pc helped.

Thanks @dimipage for you commnet. I connect it directly to the computer. What version of QMK did you use?

dimipage, 2024-11-12:
> > Do you use a usb hub by any chance? I had the same issue and switching the hub or connecting directly to pc helped.
> 
> Thanks @dimipage for you commnet. I connect it directly to the computer. What version of QMK did you use?

I've downloaded the latest repo, main branch

mraspaud, 2025-03-01:
I have the same issue, and same controller. I tried a few things, but the problem is still there...