diepala/cantor #35: Problem with bootloader and possible solution

mraspaud, 2024-12-26:
I recently built the mx version of this keyboard, and I really like it, so first of all, thanks a lot for creating this project!

Now that I'm tweaking my layout with qmk, I realise that I have a lot of problems getting the controller in bootloader mode, either with the boot0 and nrst buttons or with qmk's bootmagic, in that it works once in twenty times maybe.

I used the blackpill stm32f401ccu6 v3 from ali express, which seems to be working fine otherwise.

I googled the problem and realised that I'm not the only one with it. In particular [this thread](https://www.stm32duino.com/viewtopic.php?t=1680) seems to indicate it is known and has a simple solution of connecting pin A10 to ground (explanation on why in the last post of that thread).

Hence, I soldered a wire between the ground and A10 and can confirm that setting the microcontroller in bootloader mode now works reliably every time, either with the buttons or with bootmagic.

So my request here would be if it would be possible to patch the pcb files to add a trace between ground and a10 to avoid the problem in the future altogether?

Some comments indicate that noise on A10 might be due to static, and indeed I have my keyboard on a plastic surface, so maybe I'm more prone to the problem that others, but I think it would be quite nice with this preventive measure.

Here is a picture of my quickfix:
[image]

Thanks again for the project!
Martin

svc-user, 2025-04-25:
I now have my own Cantor MX in hand, witht he Blackpill STM32F401, and upon further reading the datasheet A10 specifically does not include an internal pull-up/pull-down resistor. The QMK website says you should use a pull-up at 22k, but I am able to get into DFU mode by shorting to ground. However I'm not sure that's the best approach as the key k25, is using pin A10. More test to be done :)

mraspaud, 2025-04-28:
@svc-user Sorry I missed you messages earlier!
Indeed, a10 is not connected to anything (ie not track on the board connects to in), hence my suggestion to add a track to ground for it...